DeskPi-Team/super6c #1: Identities of chips on board and interconnects

adamfowleruk, 2022-07-23:
Hello,

I'm trying to get a Pine64 SOQuartz CM4 module working on the DeskPi Super6C. I was wondering if you could please identify the USB hub chip in use on this board, and confirm exactly how that's connected to the 2x 100 pin connectors on the first cm4 board?

The default soquartz-cm4io DTS file doesn't work with your board - some USB interfaces are disabled by default in the latest Manjaro Arm development build. I'm attempting to find differences between the two carrier boards so that the Super6C is supported.

The docs on this repo are insufficient and the images not quite detailed enough to identify the USB hub chip.

Any help/advice greatly appreciated.

adamfowleruk, 2022-07-24:
Never mind about the USB chip - I've found it. It's a Core Chip ShenZhen Co Ltd, SL2.1A 4 post high speed USB 2.0 hub controller. The PDF data sheet can be found online in Chinese and easily translated by Google Translate to English.

Any other chip IDs and information greatly appreciated.

geerlingguy, 2022-07-29:
@adamfowleruk - Just wanted to cross-post this issue (for the CM4, but similar idea): https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/425

PBXForums, 2022-08-17:
@adamfowleruk Any chance of a copy of your DTS please, I am trying to do exactly the same thing. I don't have the board yet but I do have the soqartz. I'm using Debian right now.

closed 2024-01-30 by yoyojacky