DeskPi-Team/super6c #2: Support for VLANS in network switch

Chaz6, 2022-08-18:
Is there any facility to configure VLANs for the network switch? I would like to be able to configure tagged and untagged vlans for each port.

louispool, 2022-11-10:
AFAIK you can't do this because the switch is not managed. This is the biggest disadvantage imo when comparing this board with the Turing.

Transmitt0r, 2022-12-14:
Technically the RTL8370N can be used as a managed switch. For example this switch (https://www.dlink.com/en/products/dgs-1100-08-8-port-gigabit-smart-managed-switch) uses the same chip.

damige, 2023-01-21:
I hope this will be made possible in the future board revision. As this is the only reason i have not bought this great product already.

rr4444, 2023-01-22:
But what is it you actually want to use hardware VLAN for?

I'm curious.... 

And even then, what's wrong with doing it in software? SW is probably more flexible and automatable?

Or am I missing something?

damige, 2023-01-22:
> But what is it you actually want to use hardware VLAN for?
> 
> I'm curious....
> 
> And even then, what's wrong with doing it in software? SW is probably more flexible and automatable?
> 
> Or am I missing something?

I use a firewall with multiple vlans and a physical POE switch connecting the server(which would be the super6c) a physical firewall, accesspoints and a few other IOT-like devices.

The vlans are used because i host a web-services from home, and i dont want the devices in this (DMZ) vlan to be able to access internal devices, like my storage, or IOT.

In the case of the Super6C i would dedicate a few compute units to DMZ tasks, a few to IOT tasks and a few for internal tasks.
To be able to do this (securely) i would have to configure the internal super6c switch to allocate different access vlans to each set of compute units. And have the switch tag the packets with dot1q headers when the packets leave the internal switch toward the physical POE switch.

Not using access vlans on this switch would not be as secure because when my internet facing (DMZ) compute unit gets hacked it would be able to modify itself to access all other compute units without going through the physical firewall.

Hopefully this makes the usecase clear.

rr4444, 2023-01-22:
Ahaaaaa. I wasn't thinking outside of the cluster itself, let alone from a DMZ firewall perspective...

Makes total sense.

jdimpson, 2023-08-29:
Even in a cluster configuration, it would be nice to make one CM4 be the control node that provides DHCP, NFS and other services to a private network that contains the rest of the nodes. 

You'd do this by putting the board's two external ethernet ports on a different VLAN than you put the rest of the CM4s. Then the control node gets put onto both VLANs, and routes/firewalls/NATs between the two virtual interfaces. This way the nodes of the cluster remain separate from, but still accessible by,  the external network.

It looks like it only requires two pins off the RTL switch chip, for I2C.

damige, 2024-01-29:
> If it's a home network environment, it should be considered an intranet environment. Why make it so complicated? Wouldn't it be simpler to connect a Super6c with an Ethernet interface to a manageable switch? A single Super6c can constitute a cluster environment, making it easier to manage. Why complicate things for something that could be straightforward?

It would be easier if it was just an intranet environment, it would also be easier if there were two Super6c clusters.
The usecase is however that a single Super6c gets split over two security zones (say intranet and internet). And traffic between zones needs to be strictly curated. If Vlans are not used: when an internet cm4 gets hacked and rooted, they would be able to access the intranet nodes also.

I personally don't need more then 4 compute units, but have intranet and internet needs. Therefor vlan support would be great.

yoyojacky, 2024-01-30:
From a security perspective, indeed, exposing part of the CM4 in a public network environment could pose a certain threat to other CM4 devices in the same intranet environment. However, the Super6c baseboard cannot implement the VLAN partitioning function because all CM4 devices connected to Super6c have their Ethernet connections on the same switch chip, effectively acting as if they are connected to an unmanaged switch, merely facilitating data exchange. Currently, it seems that this requirement cannot be met. We suggest that if security concerns need to be addressed, the entire Super6c should be placed in an intranet environment for management via the intranet. Each individual CM4 system should have its firewall enabled with high complexity passwords. However, for users building clusters, this might increase the management burden.

damige, 2024-01-30:
Thanks for the insight, please consider it as a feature for a future revision or version of the board.

closed 2024-02-26 by yoyojacky