DeskPi-Team/super6c #4: POE Header?

PARitter, 2022-08-25:
There are two side-by-side 4-pin headers right behind the Ethernet jacks labeled "J11" & "j13".  The are also labelled "POE1" & "POE2".  

I presume these provide raw POE voltage from the ethernet ports.

Could you please provide documentation about these headers?

yoyojacky, 2024-01-29:
No, it does not support PoE  yet.

closed 2024-01-30 by yoyojacky

deosrc, 2025-05-06:
The README is a little confusing on this. In the features list:

- External network interface POE power pin *2

Then the only other reference to PoE seems to be:

- J11/J13: PoE1/PoE2 Not available, DO NOT CONNECT!!!

I don't really know much about PoE, so maybe I'm just not understanding the bullet in the features.