DeskPi-Team/super6c #6: NVMe boot

jasonhancock, 2022-09-18:
I've flashed my CM4's with an updated boot order to boot off NVMe if present and flashed my NVMe SSD's with an image. I cannot get them to boot with the DeskPi Super6c. I see on the product page:

> You can boot the OS either from eMMC, SD Card or netboot.

Is there some technical limitation as to why it won't or cannot boot from NVMe, or do I have a mistake in my configuration somewhere?

joshuatam, 2022-10-03:
I can boot from SSD with the following steps:

Hardware requirement:
CM4 Lite 8GB RAM

1. Use a sdcard to modify CM4 default boot order (Misc utility Images -> Bootloader -> USB Boot)

[image]

2. After boot in the CM4, wait for onscreen to detection for few minutes, power off, get back the sdcard.

3. Flash the sdcard with the latest Raspberry Pi OS 64bit (2022-09-22).

4. After booting to the GUI, open "SD Card Copier" from Accessories, copy the files to SSD
- Copy from Device: Select your sdcard
- Copy to Device: Select your NVME SSD

5. After copied, shutdown and unplug the sdcard (it is optional, as the bootloader will try to find SSD to boot first)

6. After boot you will see it is using the SSD for the whole system.

[image]

aderesh, 2022-10-31:
@joshuatam , Imager's USB boot option does NOT work with CM4 modules - I made the same mistake :). It works for you because you're using CM4 Lite and NVMe by default in the boot sequence. I have CM4 Lite - it worked for me without bootloader flashes. Also, USB boot has nothing to do with NVMe boot :) 

@jasonhancock Super6C does support booting from NVMe - I had the same question :) . I assume you have CM4 with eMMC so you need to flash a bootloader to make NVMe boot first instead of SD/EMMC. Otherwise, it will always boot eMMC. Please check this answer - https://github.com/DeskPi-Team/super6c/issues/11#issuecomment-1291204295.  Note: I had issue with one NVMe -probably hardware incompatibility so maybe you have the same issue. In that issue, I mentioned NVMe models that worked for me

joshuatam, 2022-11-07:
> @joshuatam , Imager's USB boot option does NOT work with CM4 modules - I made the same mistake :). It works for you because you're using CM4 Lite and NVMe by default in the boot sequence. I have CM4 [...]

@aderesh Interesting, so you only have 1 step to flash the OS onto SSD directly and can already boot by default?

aderesh, 2022-11-09:
correct but only for CM4 Lite. For CM4 with eMMC, SD boot goes before NVMe so you have to change boot order

aregenscheid, 2023-07-24:
@aderesh @jasonhancock @joshuatam 
How did you write the OS image to the NVMe drive? I'm having a horrible time installing Kairos.  I flashed the bootloader on a CM4 Lite. I can boot off NVMe when I use the RPi Imager tool to write Raspberry Pi OS. But, when I try to write the Kairos image to NVMe it won't boot. But it will boot when I write it to an SD card. Any ideas?

yoyojacky, 2024-01-29:
> @joshuatam , Imager's USB boot option does NOT work with CM4 modules - I made the same mistake :). It works for you because you're using CM4 Lite and NVMe by default in the boot sequence. I have CM4 Lite - it worked for me without bootloader flashes. Also, USB boot has nothing to do with NVMe boot :)
> 
> @jasonhancock Super6C does support booting from NVMe - I had the same question :) . I assume you have CM4 with eMMC so you need to flash a bootloader to make NVMe boot first instead of SD/EMMC. Otherwise, it will always boot eMMC. Please check this answer - [#11 (comment)](https://github.com/DeskPi-Team/super6c/issues/11#issuecomment-1291204295). Note: I had issue with one NVMe -probably hardware incompatibility so maybe you have the same issue. In that issue, I mentioned NVMe models that worked for me

Please refer to this wiki: https://wiki.deskpi.com/super6c/

yoyojacky, 2024-01-29:
I have test it by using following steps:
1. if you have an EMMC  verion CM4 , you may need to boot from MicroSD Card first and then format EMMC and then copy system from MicroSD card to NVMe SSD by using dd command or SD card copier tool. and then, try to reboot your super6c .
2. if you have an CM4 module without EMMC on it, you can just flash the latest image to MicroSD card and then copy the image to SSD drive by using the same tool, and then do remember to set it up by using "sudo raspi-config" tool .

closed 2024-01-30 by yoyojacky