DeskPi-Team/super6c #7: CM4 MMC

fabltd, 2022-09-29:
Hi

Cannot seem to flash CM4 EEMC in Slot 1 as the rear MINI USB does not connect to the PC Slots 2 + seem to work ok?

Any thoughts?

aderesh, 2022-10-31:
Didn't get the question but I could flash bootloader on $1 (back microusb), $2(by the CM4), $3

sloppycoder, 2023-01-05:
@aderesh , could you explain how to flash the eMMC?

I just got my Super6C board and some CM4 with 8G eMMC. I put the jumper on nRPIBoot pin, and connect the power, I can't see any USB device from my Ubuntu Linux box nor my M1 Mac.

I followed the instruction at this repo. https://github.com/raspberrypi/usbboot. Unfortunately I don't have any windows machine to try.

Thanks very much in advance.

sloppycoder, 2023-01-05:
I just realized I was using the wrong USB cable. Switched to an OTG cable then it worked 😅

aderesh, 2023-01-05:
@sloppycoder I used RP4 with stock Raspberry PI OS + usbboot.
Make sure you use a proper micro USB slot that corresponds to the CM4 slot: microUSB for CM4 $1 slot is on the I/O/back panel, others microUSBs are at the corresponding boards. The order of CM4 on the board is finicky - make sure you find a marking like $1, $2 etc on the back of the board. 
You mentioned that you put a jumper on nRPIBoot pin, that's correct and is required:
[image]

I don't remember if USB was recognized - I do assume so :) I followed this guide - https://www.jeffgeerling.com/blog/2022/how-update-raspberry-pi-compute-module-4-bootloader-eeprom. 

Note: I have CM4 without eMMC but it should not make a difference as far as I know.

DonBower, 2023-04-10:
> @sloppycoder I used RP4 with stock Raspberry PI OS + usbboot. Make sure you use a proper micro USB slot that corresponds to the CM4 slot: microUSB for CM4 $1 slot is on the I/O/back panel, others [...]

This worked for me.

yoyojacky, 2024-01-30:
For example. 
1. you have a 32GB EMMC storage on CM4 board. 
2. Prepare a 16GB TF card with the latest Raspberry Pi OS (32bit or 64bit ) . 
3. Follow the instruction at this repo. https://github.com/raspberrypi/usbboot.  ( In my case, I use a raspberry Pi 4B as my pc ) 
4. Put a jumper cable on nRPIBOOT and GND pin on Super6c CM4 Jumpers.  it will disable booting from EMMC. 
5. Insert the TF card to TF  card slot. 
6. Connect the OTG wire ( MicroUSB cable with data lines) to MicroUSB cable, USB2.0 port to your Linux system or your PC  ( In my case, I use a raspberry Pi 4B as my pc ) which has been deployed usbboot. and please do remember execute `sudo  ./rpiboot ` before connect the power. 
7. Connect the power to super6c, you will find that the EMMC will be recognized as a `sda` device  on you linux system.  you can use `dd` command to copy the image from TF card to EMMC by following command:
```bash
sudo dd  if=/dev/mmcblk0  of=/dev/sda bs=4M status=progress 
```
After copying. turn off the power, remove the jumper from `nRPIBOOT and GND` and then remove the TF card, just reconnect the power back, it will start booting from EMMC.

closed 2024-01-30 by yoyojacky