DeskPi-Team/super6c #11: Boot from NVMe (no SD, no USB, no NETBOOT)?

aderesh, 2022-10-25:
I have CM4 Lite (no eMMC) at $1 and a 512 GB NVMe. If I boot from SD card with Raspbian, I can see and use the attached NVMe. However, if I install raspbian on NVMe and try to boot using NVMe only, I see only a black screen. If I remove all storage devices, I can see that the PI cycles through USB-MSD, BCM-USB-MSD,  PCI/nvme and finally Network boot (0xf25641) so the NVme is definitely in the sequence. I'm planning to upgrade my CM4 bootloader from 507b2360 2022/04/06 to the newest version to see if it helps, but does the board support booting from NVMe?

aderesh, 2022-10-25:
If not, is there a plan to support NVMe in the future?

avluis, 2022-10-25:
I can say for certain it does support booting from NVMe; caveat in my case is that all my boards are populated with eMMC; but:
```shell
root@clpi-cm4a ~# neofetch
       _,met$$$$$gg.          root@clpi-cm4a
    ,g$$$$$$$$$$$$$$$P.       --------------
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 11 (bullseye) aarch64
 ,$$P'              `$$$.     Host: Raspberry Pi Compute Module 4 Rev 1.0
',$$P       ,ggs.     `$$b:   Kernel: 5.15.68-v8+
`d$$'     ,$P"'   .    $$$    Uptime: 3 hours, 10 mins
 $$P      d$'     ,    $$P    Packages: 875 (dpkg)
 $$:      $$.   -    ,d$$'    Shell: fish 3.5.1
 $$;      Y$b._   _,d$P'      Terminal: /dev/pts/0
 Y$$.    `.`"Y$$$$P"'         CPU: BCM2835 (4) @ 1.500GHz
 `$$b      "-.__              Memory: 595MiB / 3844MiB
  `Y$$
   `Y$$.
     `$$b.
       `Y$$b.
          `"Y$b._
              `"""

root@clpi-cm4a ~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0  29.1G  0 disk
├─mmcblk0p1  179:1    0   256M  0 part
└─mmcblk0p2  179:2    0  28.9G  0 part
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
nvme0n1      259:0    0 476.9G  0 disk
├─nvme0n1p1  259:1    0   256M  0 part /boot
└─nvme0n1p2  259:2    0 476.7G  0 part /
root@clpi-cm4a ~#
```

Here is what I edited into ~/usbboot/recovery/boot.conf (cloned usbboot from: https://github.com/raspberrypi/usbboot and pulled latest firmware while at it from: https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/stable)

```shell
root@clpi-cm4a ~# cat ~/usbboot/recovery/boot.conf
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
#BOOT_ORDER=0xf25641

# Try NVMe PCIe first (6), followed by, SD, USB PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25416

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```
Found this that can walk you thru this if needed: https://jamesachambers.com/full-compute-module-4-raspberry-pi-setup-imaging-guide/

aderesh, 2022-10-26:
@avluis appreciate the confirmation and the guides! 
It turned out to be some sort of hardware incompatibility with the Micron 512 GB I took from a Dell laptop after the upgrade. Flashing the newest stable bootloader didn't help either - black screen and constant green LED. It has been working as a storage with a bootable SD card. So, I ordered a MMOMENT NVMe M.2 2280 PCIe Gen 3x4, Solid State Drive Internal SSD (MT34 256GB) for a test and it worked! :) 

I can see that you have 512Gb working, I'll test 1TB tomorrow :) 

It does boot from USB drive as well so I guess all storage types are supported - USB drive for $1 with netboot for $2-$6 was my backup plan and I'm glad that I don't have to go that route (unless I want to try netboot :)

closed 2022-10-26 by aderesh