DeskPi-Team/super6c #13: Is it possible to mount an SD Card with a CM4 with eMMC?

louispool, 2022-10-30:
Hi there, as far as I am aware you **cannot** mount an [SD Card to a CM with eMMC](https://www.reddit.com/r/retroflag_gpi/comments/snesyy/is_it_impossible_to_mount_the_sd_card_with_an/). 

Even though it's not explicitly stated, the documentation on the Super6c github somewhat misleadingly states in the [TroubleShooting Section](https://github.com/DeskPi-Team/super6c#troubleshooting) that
	> "If your CM4 module has eMMC on board, the SSD drive **and TF card** can be external mass storage." 

I fitted an SD Card in the TF Slot of my CM module with eMMC and could not find the device as external mass storage. Am I missing something? Or is this a mistake in the documentation?

avluis, 2022-11-23:
Believe the documentation is wrong here as there is no way to do so unless the board re-wires the relevant pins and provides respective overlays to enable this (this is for example with SPI)

TL;DR no, it is not possible with this combination

yoyojacky, 2024-01-30:
> Hi there, as far as I am aware you **cannot** mount an [SD Card to a CM with eMMC](https://www.reddit.com/r/retroflag_gpi/comments/snesyy/is_it_impossible_to_mount_the_sd_card_with_an/).
>
> Even [...]

Let me see, your super6c is booting from EMMC onboard, right?  Could you please try to execute this command :
```bash
lsblk 
``` 
and send the feedback to me ?

closed 2024-07-05 by yoyojacky