DeskPi-Team/super6c #14: Do 3 12v 4-pin fan support PWM?

aderesh, 2022-10-31:
It seems like they don't because my goes at max speed all the time despite the temperature. Just want to confirm

avluis, 2022-10-31:
Seeing the same behavior; would need to prove i2c (as well as checking schematics if any to be certain)

karhohs, 2023-11-07:
Hello, I see that the included fans do not support PWM. As a follow up question, is there a way to control whether a fan is on or off from a CM4? If so, how? Thank you!

karhohs, 2023-11-14:
I did ultimately replace the stock fans, which are very loud, with Noctua fans, specifically the nf-a4x20-flx. The difference in ambient noise is stark and whisper quiet with the new fans. These are not PWM fans, but they include low-noise adapters to adjust the fan RPM if desired. I wanted to use the same mounting screws to attach these fans, so I had to create a kind of adapter to get the screws to fit to the fan; here is a link if interested, https://www.printables.com/model/644669-adapter-for-replacing-stock-fans-with-noctua-in-de.

yoyojacky, 2024-01-30:
Sorry, It dose not support PWM.

closed 2024-01-30 by yoyojacky