DeskPi-Team/super6c #15: Reliably start all Pis

Asara, 2022-11-23:
This is a bit of a follow up to: https://github.com/DeskPi-Team/super6c/issues/10

Instead of powering up a single pi after shutdown, I am wondering if there is anything i can do to reliably get all of the CM4 units to boot up from cold boot. Or, on the flip side, force a pi to reboot individually. 
On a cold boot I typically see between 3 to 5 Pis come up, with the others coming on but never booting (red LED never goes off). Is there any way to force these Pis to restart?

I am using the provided 100W power supply. Using 6 CM4 Lites with 6 SD cards and NVMe drives attached to each Pi if that might effect it.

Would love some advice here, and thanks!

avluis, 2022-11-23:
This sounds like hardware or power overdraw; can you list the NVMe's that you're using (model names if possible)
CM4s are powered on sequentially which actually helps with power-on load however minimal it is but accessories (NVMe, SD Cards) can affect this.
Which module position (if consistent) does not come up?
Also; have you double checked you have a good image?

Asara, 2022-11-23:
Might have edited my original post a bit too late. 
I've tried removing the NVMe drives as well, and still don't get all the modules to come up.

When all the modules do come up (which has happened with all the NVMe drives attached), they work fine, am able to put some stress on all the CPUs at the same time and not have an issue (ran apt upgrade across all 6 at once with no issues).

Model of the NVMe: WDS100T3G0C
Haven't actually mounted any of these yet so right now its just running off the SD card, which is required since I have the CM4 Lite.

From the last 4 boots, module #4 has been the one to consistently not come up (both with and without the NVMe).  On my last reboot it was the only one not to come up.

avluis, 2022-11-23:
Two tests:

 - Swap the SD card in slot 4 to slot 1 and note what slot fails to come up -- keep that SD card in slot 1 for the next test

 - Swap the CM4 in slot 4 to slot 1 and see if the same behavior persists in slot 4 (w. SD Card from slot 1) or slot 1 (w. SD Card from slot 4)

Asara, 2022-11-23:
I've mixed and matched SD cards + compute modules, and there doesn't seem to be any correlation between starting and the SD card, nor the compute module.

When testing with the swaps between Module 1 and 4, suddenly module 3 and 5 wouldn't start.

It is strangely intermittent. Currently the board is on with all 6 compute modules running, with all 6 NVMe drives attached. No messages about low voltage in dmesg on any of the drives either.

avluis, 2022-11-23:
Good info; I would look into replacing the board unless you are able to test each module + SD card individually (separate from each other on another carrier board)