DeskPi-Team/super6c #17: Use microusb ports as UART?

sloppycoder, 2023-01-05:
Is it possible to use the micro USB ports as UART console? If so, what type of adapter do I need to get? And what configuration is needed? 

Thanks very much in advance.

avluis, 2023-01-08:
You won't be able to use an overlay to do so as there are no exposed GPIO -- you would need to look for an appropriate USB to UART/TTL Serial adapter for your needs (check Adafruit)

rr4444, 2023-01-14:
> You won't be able to use an overlay to do so as there are no exposed GPIO -- you would need to look for an appropriate USB to UART/TTL Serial adapter for your needs (check Adafruit)

Hi, I guess you are meaning about how to connect to the first CM4 via tty? 

Instead, what I think the OP was hoping for was the ability to access the UARTs for serial console on all the CM4s i.e. not just the one with the exposed USB

rr4444, 2023-01-14:
That type of serial console access is really important for remote/edge type maintenance if networking doesn't come up properly. 

My personal hope was to have another rpi/one of the six rpis able to access the others' serial consoles.

rr4444, 2023-01-14:
Actually, just dawned on me that the Bluetoothed CM4s could talk to each other as TTYs...

I guess?

(I am considering the Super6c and am waiting my first CM4s...)

yoyojacky, 2024-01-29:
There are several pins of GPIO has been spread out, you can refer to the wiki:https://wiki.deskpi.com/super6c/

closed 2024-01-29 by yoyojacky