DeskPi-Team/super6c #23: How to power cycle individual CM4?

Gooseman42, 2023-04-04:
I would assume that there is some way to access the power management IC STC15W408AS via I2C from any CM4 slot (similar to the Turing Pi) to power cycle any individual module?

bigfish24, 2023-07-04:
Also interested in this for a chaos testing setup.

derzahla, 2023-07-18:
Came here for an answer to this question. Feel unsatisfied.

diyoyo, 2023-09-21:
I would like to learn more about this too.

boldandbusted, 2024-01-15:
Maybe opening up the GitHub Discussions would benefit the project? :)

tltangliang, 2024-01-17:
I'd suggest they go with the rp2040 as the power controller for the next generation, so there's enough playability.

yoyojacky, 2024-01-29:
As I know, i do power cycle individual CM4 by using Ansible script on software, it can not do power cycle in hardware way, because all of the Raspberry Pi CM4 are connect to power core at the same time.

yoyojacky, 2024-01-29:
> I'd suggest they go with the rp2040 as the power controller for the next generation, so there's enough playability.

Good Idea, you mean using a Raspberry Pi PICO as power controller ? it sounds interesting... LoL,

closed 2024-01-30 by yoyojacky