DeskPi-Team/super6c #25: Can not get USB to work (it works with CM4 in a regular IO board)

artlav, 2023-05-12:
I got a Super6C board and can't get the USB of the cm4 in slot 1 to work.
Verified it works with the cm4 on a regular io board, tried every other config.txt setting i could find being suggested, nothing.

lsusb does show a root hub, but nothing plugged into any port is detected.

The ports have 5V in them, there is nothing in microusb slot.

I'm using 64bit system, on an 8Gb RAM, 32Gb EMMC, with wifi CM4.

artlav, 2023-05-17:
Follow-up: I can occasionally get it to work by touching and/or flexing the board around the # 1 cm4's microusb port (the one sticking out the back).

Can't quite tell if there is some pull up or pull down resistor missing, or if it's some sort of a connectivity issue in the PCB.

Once triggered, USB keeps working through reboots, but goes back to being broken once power is cycled.

Primary suspects for now either 5V on that needs to be pulled to ground, or ID pin needs to be pulled somewhere.

Would any of this make sense?

Anyone else had the same problem?

artlav, 2023-05-17:
Tried pulling down ID pin and pulling down 5V pin (with 10k resistors), no effect.

The problem remains intermittent - about every 10th boot results in working USB, and apparently touching the board does not make a difference.

The state of it is visible as early as the bootloader, so it's not a linux issue.

yoyojacky, 2024-01-29:
You can add this parameters to /boot/config.txt  and reboot your Raspberry Pi and try again: 
 dtoverlay=dwc2,dr_mode=host

yoyojacky, 2024-01-29:
https://www.raspberrypi.com/documentation/computers/compute-module.html#compute-module-io-board-for-cm4  Please refer to this URL, It is the same as the official I/O board 's  functions.

closed 2024-02-02 by yoyojacky

binaryben, 2024-10-16:
I'm having the same issue. Can't get the USB A ports to work. I already have tried manually setting `dtoverlay=dwc2,dr_mode=host` and removing `otg_mode=1` with no difference.

```
pi@raspberrypi:~ $ dmesg | grep -i usb
[    0.084808] usbcore: registered new interface driver usbfs
[    0.084866] usbcore: registered new interface driver hub
[    0.084930] usbcore: registered new device driver usb
[    1.796018] usbcore: registered new device driver r8152-cfgselector
[    1.796075] usbcore: registered new interface driver r8152
[    1.796119] usbcore: registered new interface driver lan78xx
[    1.796162] usbcore: registered new interface driver smsc95xx
[    1.797193] usbcore: registered new interface driver uas
[    1.797242] usbcore: registered new interface driver usb-storage
[    1.804265] usbcore: registered new interface driver usbhid
[    1.804284] usbhid: USB HID core driver
[    2.490274] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    2.490582] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    2.545178] dwc2 fe980000.usb: DWC OTG Controller
[    2.545253] dwc2 fe980000.usb: new USB bus registered, assigned bus number 1
[    2.545325] dwc2 fe980000.usb: irq 37, io mem 0xfe980000
[    2.545590] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.06
[    2.545627] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.545650] usb usb1: Product: DWC OTG Controller
[    2.545667] usb usb1: Manufacturer: Linux 6.6.31+rpt-rpi-v8 dwc2_hsotg
[    2.545685] usb usb1: SerialNumber: fe980000.usb
[    2.547469] hub 1-0:1.0: USB hub found
[    6.565930] usbcore: registered new interface driver brcmfmac
```

Any chance this is indicative of a hardware issue maybe?

```
[    2.490274] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    2.490582] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
```

I'm curious why the OTG port still works but the switch isn't handing over to the hub like it's supposed to?

yoyojacky, 2024-10-21:
These two lines of log information come from the kernel log of the Linux operating system and are related to the initialization of the USB controller (dwc2). The dwc2 is the driver for Texas Instruments' USB 2.0 controller, commonly used in embedded systems and single-board computers, such as the Raspberry Pi.

The log messages "supply vusb_d not found, using dummy regulator" and "supply vusb_a not found, using dummy regulator" indicate that the kernel did not find power supplies named "vusb_d" and "vusb_a" during the initialization of the USB controller. In embedded systems, power supplies are typically used to provide stable voltage to hardware devices. If the system is not correctly configured with these power supplies, the driver will use a dummy power supply as a substitute to ensure that the USB controller can function properly.

This is usually not a serious issue because the dummy power supply can act as a fallback solution to ensure that the USB controller can initialize even if the power supplies are missing. However, if USB devices require specific voltage or current, using a dummy power supply might lead to performance issues or devices not working properly.

If you encounter problems when using USB devices, you may need to check your hardware design or power management configuration to ensure that all power supplies are correctly set up. If you are developing or debugging an embedded system, these log messages can serve as clues for diagnosing issues.

binaryben, 2024-10-24:
> If you encounter problems when using USB devices, you may need to check your hardware design or power management configuration to ensure that all power supplies are correctly set up. If you are developing or debugging an embedded system, these log messages can serve as clues for diagnosing issues.

I'm unsure what you're referring to here in terms of checking my hardware design? I'm plugging in off the shelf keyboards, USB hubs (powered and unpowered), flash drives, etc. None are working. I've tried using the micro port with an OTG adapter and the built in host ports. As far as I am concerned, it is the Super6c that likely has the hardware issue if anything does. Thank you for the rest of your message - I have been researching a lot since my last message to understand USB implementations more. It helps confirm what I've learned.

***

### Reverse Engineering / Debugging so far

It appears the OE (Output Enable) pin (10) of the USB 2.0 multiplexer IC (RS2227) is HIGH (using 3v3 logic based on V+ connected to the IC). The micro port is connected to HSD2. HSD1 is routed under the RS2227 IC to the USB 2.0 hub IC (CoreChips SL2.1A TT0122A21). With nothing else plugged in, the Select pin (2) is HIGH which is as expected for the USB micro port to work.

What is unexpected though is that the 5V pin of the OTG port is not powered[^otg] and that the OE pin is HIGH. The Function Table on [page 2 of the datasheet](https://xonstorage.blob.core.windows.net/pdf/runic_rs2227xn_apr22_xonlink.pdf) indicates it should be low for the mutliplexer and either port to be active. Unfortunately the OE trace goes straight into a via and I'm unsure where it comes back up, or if it's just connected directly to a power layer on the PCB? The fact it is high explains why dmesg has nothing to report when anything is plugged in - the multiplexer has all the downstream ports disconnected.

> The ports have 5V in them, there is nothing in microusb slot.

Can confirm this as well. USB A ports and internal headers appear to be powered directly from internal 5V rail - no relation to any IC driven logic.

> lsusb does show a root hub, but nothing plugged into any port is detected.
> 
> I'm using 64bit system, on an 8Gb RAM, 32Gb EMMC, with wifi CM4.

Same module. Same issue - but I don't think it's specific to the module variant. 

> Can't quite tell if there is some pull up or pull down resistor missing, or if it's some sort of a connectivity issue in the PCB.
> 
> Primary suspects for now either 5V on that needs to be pulled to ground, or ID pin needs to be pulled somewhere.

Leaning to the same conclusion. Should the OE pin on the multiplex be tied to ground instead of 3v3? Otherwise how is the logic of the select and OE pins on the RS2227 supposed to be triggered in normal operation?

[^otg]: I assume the 5v pin on the micro isn't powered because it's expected that this is a power input when flashing the eMMC, but even adding a powered USB hub via an OTG adapter isn't being detected.

binaryben, 2024-10-24:
Yeh, I plug in a standard micro USB cable into the slot 1 usb micro and a host computer (as if to flash it, but whilst the system is powered), the select pin goes LOW. The logic is the wrong way around, especially if this comment is indicative of the intention of the board design:

> https://www.raspberrypi.com/documentation/computers/compute-module.html#compute-module-io-board-for-cm4 Please refer to this URL, It is the same as the official I/O board 's functions.

Please see the schematic of the official CM4 IO Board below. R11 would normally pull the select pin LOW in normal operation, and then R8 should pull it HIGH when an external host is connected (via the hosts 5V over USB), but the opposite is happening (at least on my Super6c).

In addition, the OE pin is HIGH when it should be LOW. Nothing I'm doing during debugging seems to affect this so I am assuming it's just connected directly to one of the inner power PCB layers with that via. If so, this is a design mistake though compared to the official board. Pin 10 of U13 on the official schematic shows it should be tied directly to GND. This issue should be reopened and investigated properly please.

[image]

binaryben, 2024-11-05:
Hello, has anyone seen this? @yoyojacky?

yoyojacky, 2025-02-06:
> Hello, has anyone seen this? [@yoyojacky](https://github.com/yoyojacky)?

Hi binaryben, 
Did you check the I/O board 's schametics ? It is the same as the super6C does,  if you want to use MicroUSB port ,you may need to add `dtoverlay=dwc2` to /boot/firmware/confit.txt file and reboot your device ,and at the same time ,if you want to update the eeprom via USB port ,you may need to install the usbboot via [RPi-Official-repo](https://github.com/raspberrypi/usbboot) to boot from USB and flash the image to eeprom.