DeskPi-Team/super6c #29: Compatible with Radxa CM5

tburschka, 2023-09-05:
Hi,

The module isn't released yet, but can you tell based on the specs, if it could be working?

https://www.cnx-software.com/2022/09/08/radxa-cm5-rockchip-rk3588s-module-raspberry-pi-cm4/

yoyojacky, 2024-01-29:
Hi, It is for Raspberry Pi CM4 Only.

yoyojacky, 2024-08-21:
https://www.youtube.com/watch?v=-xSTThgtFhY  I've noticed that this man had put a BananaPi CM4  in to the board and works well.

gutleib, 2024-11-10:
Also, very interested in compatibility of this combo

yoyojacky, 2025-02-06:
> Also, very interested in compatibility of this combo.
> 
> PS Radxa list it as compatible, but without any details. If you could confirm it, it'd help

It seems like it can not work properly as Raspberry Pi CM5, I have no idea about the Radxa CM5 on super6C, it may be as same as the Raspberry Pi CM4.

sgsaenger, 2025-02-10:
I can confirm the Radxa CM5 to properly boot and access the network. It can also be flashed inside the Super6c. Did not try NVMe or HDMI so far, though.