DeskPi-Team/super6c #35: CM4 in #1 now working

youngjin-yun-kurly, 2024-05-10:
I bought a new one yesterday and booted it up with the CM4 in #1, but there is no HDMI, USB, or NVME access. 

I managed to get the IP on the network and it works, but when I checked the status with commands like lsusb, lspci, lsblk, nvme, etc, the resources don't exist. 

What should I do?

yoyojacky, 2024-07-05:
Hi @youngjin-yun-kurly ,
what kind of CM4 did you use?  and what OS did you use? 
Could you please tell me more details of your operation ?

lamedts, 2025-06-20:
i just got my new super6c, and got the same issue. cm4 #1 slot is not working.
i am using raspberry pi cm4, 
I tried to connect via HDMI and got a blank screen after rainbox screen 
but if I put the cm4 into other slot, it works fine
@yoyojacky